eclipse-opensovd/classic-diagnostic-adapter #27: Support GET /components/{ECU}/configurations/{service}

theswiftfox, 2025-08-15:
Currently only PUT is supported in `/components/{ECU}/configurations/{service}`.
Adhering to the SOVD spec we need to support GET as well.

theswiftfox, 2025-10-06:
Sure, I assigned you!
You might need to check, if the demo ECU in https://github.com/eclipse-opensovd/classic-diagnostic-adapter/tree/main/testcontainer has the required test data.

RaviprasadBS, 2025-10-07:
@theswiftfox  Thanks. Sure, will let you know if i need more info on this.

theswiftfox, 2025-11-19:
@RaviprasadBS 
Do you want to start working on the part with configuring the mapping and then checking the services against it? That should already be possible with the current open source diagnostic description I think.

alexmohr, 2025-12-01:
The existing MDD contains 0x2E for vindata indentfier

[image]

and a couple of 0x22 services. 
[image]

So from MDD side you should be good to go. 
As you're testing with the ECU sim anyway I think it would make sense to write an integration test right away. 
See https://github.com/eclipse-opensovd/classic-diagnostic-adapter/blob/main/integration-tests/tests/sovd/ecu.rs and https://github.com/eclipse-opensovd/classic-diagnostic-adapter?tab=readme-ov-file#integration-tests 

Let me know if you need support writing these tests, as you will be the first aside implementing one. :)

alexmohr, 2025-12-02:
Technically the integration test utils should contain everything you need.
Take a look at `test_ecu_session_switching`. This test is sending POST/GET, controlling the sim etc, this should be pretty similar to the test you'll have to write. 
 
`integration-tests/util/http.rs` contains helper functions to send and receive data via reqwuests. 
`send_cda_json_request` might be a helpful for example. 

The main project readme contains documentation on how to run the tests in general (linked in my last comment).

If you want we can discuss in more detail on slack.

theswiftfox, 2026-10-05:
Closing as completed with #176

closed 2026-10-05 by theswiftfox