eclipse-opensovd/classic-diagnostic-adapter #64: /modes/security assumes specificly named data

floroks, 2025-09-22:
The `/modes/security` requires the diagnostic descriptions parameter for the data to be sent as payload to be named  `Send_Key`. 

We should have a generic "parameters" key in the payload which is then provided to the send_key service as input parameters.

Example:
```
{
 "value": "Level_5",
 "parameters": {
    "SecurityKey": "0x10 0x20 0x30 0x40"
  }
}
```

alexmohr, 2026-08-27:
@theswiftfox I guess we can close this once your proc macro PR (#427) is merged?

theswiftfox, 2026-10-05:
partially. that PR so far only introduced the overwrite for the lookup itself. to have it fully dynamic and overwriteable we ideally would need to make the whole handling inside the put overrwriteable, or hand the payload in a generic way to the overridable function. As of now it still makes some assumptions i think.